ros-controls/control_toolbox #34: Integrator Windup

paulbovbel, 2015-03-18:
Even though `i_term` is clamped to `i_max`, `i_error` will continue to windup as long as there's an error. I've found this troublesome in an application with quadrotor control, where I'm trying to migrate the bespoke PID methods to the control_toolbox ones.

Would you be open to a pull adding a "antiwindup" parameter to enable clamping `i_error` by `i_max/i_gain_`?

adolfo-rt, 2015-04-01:
Are you thinking of this as an opt-in behavior, or as an always-in addition?.

paulbovbel, 2015-04-01:
Opt-in

adolfo-rt, 2015-04-01:
OK, if you can prepare a pull-request we can review it. So there would be a boolean parameter for toggling the anti-windup behavior?.

paulbovbel, 2015-04-01:
I think that would make sense, yes.

ipa-mig, 2016-06-23:
Seeing that #52 has been merged, should this be closed?

Did you change the API of `initPid` on purpose? In the constructor you added the default argument for windup to `false`, thus keeping the old behaviour. Wouldn't it make sense to do this for the `initPid` calls the same way?

(Yeah, I know I'm about six weeks late ;-) )

paulbovbel, 2016-06-23:
Thanks @ipa-mig, it would have made perfect sense to modify initPid the same way.

closed 2016-06-23 by paulbovbel

ipa-mig, 2016-06-24:
@paulbovbel will this be done for others not stumbling upon the same? (Even though I actually appreciate having a proper antiwindup and would have probably not found this change without the build error :-) )

paulbovbel, 2016-06-24:
I'm totally open to it, although @bmagyar would be the one to ask. Would you be able to put up a PR?