ros-controls/control_toolbox #35: Possibility to choose different type of controllers?

WG-, 2015-04-22:
As far as I read it is not yet possible with ros-controls to define different controllers other than a PID controller, that is to say, you can not choose for example for an lowpass filter in combination with a notch filter (or lead-lag filter, or any other standard type of filters) and choose different discretization schemes, am I correct? Is this because no one needs them? Otherwise, I might be willing to start implementing these kind of controllers.

davetcoleman, 2015-04-22:
I believe it just hasn't been done, and new control methods would be very welcomed. Note that one of the design decisions of ros_control has been to not "chain" controllers in the way that e.g. OROCOS does, but rather if you want to chain controllers you need to put each piece within the same controller. This makes ros_control much simpler than the complex OROCOS, but is frequently all that is needed

adolfo-rt, 2015-04-23:
The filters you mention have not been implemented because nobody has had the need for them. I haven't looked into it, but I'd expect there to be well tested implementations with permissive licenses out there. If this is the case, it would make sense to leverage one such implementation and expose convenient ROS bindings to it, like configuration through ROS params and dynamic reconfigure (as is done for the PID class).

On chaining controllers and filters, let's keep an eye on the future, as work done today will likely find good use in the next major iteration of `ros_control`, which I'm hoping will take place at some point in time.

WG-, 2015-04-28:
Well my guess that the need has not been there yet has something to do regarding the fact that most applications do not run on a high sampling frequency and have a low bandwidth. However, if you want to have an high performance system with a high sampling frequency and high bandwidth you will need such controllers.

Anyhow, question @davetcoleman how do you mean to not "chain" controllers? Can you give an example? I am not familiar with OROCOS namely. How I understand chaining. For instance, you have two lowpass filters H1 = 1/(s + 1) and H2 = 1/(s + 2) you will not use y = H1*H2*u (u is the input, y is the output) but rather y = H*u, where H = 1/(s^2 + 3*s + 2)?

davetcoleman, 2015-04-29:
OROCOS lets you take various components and rearrange them in arbitrary re-usable configurations is all I was saying. I thought chaining is what you meant when you said "combining filters"

christophfroehlich, 2025-04-02:
This repository is not directly related to ros2_controllers, but ros2_controllers could be chained arbitrarily. 
Filters in this repo can also be put in a filter chain.

Closing this issue as outdated.

closed 2025-04-02 by christophfroehlich