ros-controls/control_toolbox #40: DANGEROUS: Wrong behavior at startup after 1.13.0 -> 1.13.1 upgrade

marcoesposito1988, 2015-05-22:
After upgrading today I started my robot (KUKA LWR4+ with effort joint_trajectory_controller and MoveIt) and it immediately started going at full speed without any reason. I could observe this behavior consistently until I downgraded control_toolbox to 1.13.1 (it tried to reach the same position again and again at full speed, without any care for the joint limits).

This is very dangerous (I had my hand on the red button by pure chance, otherwise I would have damaged my robot very heavily).

Am I the only one with this problem? 

@carlosjoserg, I see there is a commit of yours there. Didn't you have any problems with this? Strange, since we have the same setup.

adolfo-rt, 2015-05-22:
Thanks for reporting. I believe to have found the culprit. A sign was [mistakenly flipped](https://github.com/ros-controls/control_toolbox/commit/bc45970b685c3cfdb33672897a3eec090b28ee71?diff=unified#commitcomment-11329517) in bc45970b685c3cfdb33672897a3eec090b28ee71. I believe you must be using this method.

A fix was proposed in #39. I'll roll out 1.13.2 right away, and a communication to the relevant mailing lists.

adolfo-rt, 2015-05-22:
1.13.2 released into [indigo](https://github.com/ros/rosdistro/pull/8520) and [jade](https://github.com/ros/rosdistro/pull/8521).

closed 2015-05-22 by adolfo-rt

marcoesposito1988, 2015-05-22:
Bingo! It works now. Thanks a lot for the quick response!

adolfo-rt, 2015-05-22:
Thanks for confirming the fix.

carlosjoserg, 2015-05-22:
> Didn't you have any problems with this? Strange, since we have the same setup.
No, I was still using 1.13.0 from synaptics.

Certainly quick, I couldn't even blink!

rctoris, 2015-05-22:
@tfoote Any chance we can get a new sync given this critical bug now in the debians?

tfoote, 2015-05-22:
Yeah, once things have finished building. The farm has a queue of ~ 3700 packages to build at the moment.