ros-controls/control_toolbox #79: Port to ROS2 Dashing

sloretz, 2019-05-22:
I'm looking into how much work it would be to port this package into ROS 2 Dashing (looks like `realtime_tools` needs to be ported first).

* Is someone already working on this?
* Where should a ros2 port live? How about a `dashing-devel` branch in this repo?
* Would the maintainers be willing to release into Dashing once code has been ported?
* Should the existing code style be maintained, or should the [ROS 2 style guide](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#id1) be used?
* Should deprecated methods be kept or removed?
  * https://github.com/ros-controls/control_toolbox/blob/87c3ca1efedaa5740b517ac706c27d56ee6b3559/include/control_toolbox/pid.h#L329

bmagyar, 2019-05-23:
Hello, 

For most: https://github.com/ros-controls/realtime_tools/issues/35

Deprecation: yes it is easily time to drop that.

DLu, 2021-04-30:
This can probably be closed

bmagyar, 2021-04-30:
Good point!

closed 2021-04-30 by bmagyar