ros-controls/control_toolbox #82: Cannot set PID gains fast

azompas, 2019-06-18:
Hello,

I am working on a some kind of adaptive PID control strategy where I need to set the PID gains fast. However the update of the dynamic reconfigure takes too much time, making my controller to not behave as it should.

https://github.com/ros-controls/control_toolbox/blob/kinetic-devel/src/pid.cpp#L241

Would it be possible to provide a function without the update dynamic reconfigure functionality?

Thank you.

bmagyar, 2019-06-19:
Are you sure this is happening because of the reconfigure step and not the mutex-protected realtime publisher?

azompas, 2019-06-20:
Yes, I commented out the dynamic reconfigure and my controller works as expected.
Would it be possible to have an option to disable the reconfigure part?
Or, if you are willing to merge, we could write a simpler class without the dynamic reconfigure part and make a pull request.
I believe that this functionality will be useful to anyone who wants to create an adaptive PID controller.

bmagyar, 2019-07-02:
I would prefer if you could implement this by introducing a new flag to the class named `disable_dynamic_reconfigure` and have it set to true by default (so that we don't change the existing default behaviour).