ros-controls/control_toolbox #85: Refactor the Pid class to be completely ROS agnostic

bmagyar, 2020-03-25:
...and create a new RosPid in a separate package that offers parameterization and services through the node interface. This would wrap the features now offered between the pid_gains_setter and the stuff coupled inside Pid::init*() functions

Split from #83

ddengster, 2020-04-01:
As I mentioned in the PR, should we put the parameter 'i_clamp' in?
https://github.com/ros-controls/control_toolbox/blob/melodic-devel/src/pid.cpp#L126

Reasoning being:
> well someone who had the i_clamp parameter and was using it in ros1 would need to substitute it for 2 more parameters when porting to ros2 - and tracking it down would not be very obvious especially when we don't generally have the time to put up changelists. Also I think we can spare the few lines of code.

matthew-reynolds, 2020-04-02:
> should we put the parameter 'i_clamp' in?

Just my 2 cents, seems redundant to me and not worth supporting for such a minor migration difference (ROS1=`i_clamp` -> ROS2=`i_clamp_min` and `i_clamp_max` is pretty painless).

ahcorde, 2020-06-18:
PR https://github.com/ros-controls/control_toolbox/issues/85

jordan-palacios, 2020-09-02:
@bmagyar As @ahcorde pointed out, I think we can close this now thanks to the work done in #97.

bmagyar, 2020-09-08:
Agreed, thanks!

closed 2020-09-08 by bmagyar