ros-controls/control_toolbox #151: Change in installed headers between version 2.1 and 2.2

michalpelka, 2023-03-02:
We are using "control_toolbox" as a library in our project and we experience unexpected compilation errors between version 2.1 and 2.2 in ubuntu 22.04.

Basically due to a change in "CMakeList.txt"  in `install` section introduced in commit : https://github.com/ros-controls/control_toolbox/commit/9faaa68b27c6b4b4255fe8ad977780234a8e6131 
the `control_toolbox` headers files are installed in `/opt/ros/humble/include/control_toolbox/control_toolbox/` for version 2.2.0 and for recent version 2.1.2 was installed `/opt/ros/humble/include/control_toolbox/`. 
Is this change intentional?

**Context**
The original issue is here: https://github.com/o3de/o3de-extras/issues/198.

bmagyar, 2023-03-02:
Yes it is. This enables working with the package in a workspace overlay while having the upstream package installed too. Yes it is not an elegant solution but the only remotely civilized one. It should not affect existing codebase when used with colon/ament unless you directly reference the filepath which has changed.

If you are interested please follow up with the references in the original PR.

michalpelka, 2023-03-15:
We managed to resolve the issue on our side. Thanks for the helpfull explanation @bmagyar !

closed 2023-03-15 by michalpelka