ros-controls/control_toolbox #154: Let's add a bit of documentation

christophfroehlich, 2023-03-13:
- The very first line in the readme of the ros2-master branch points to non-ROS 2 docs
- Maybe add/link some demos as proposed in #70. I know that one can use the tests to figure out how to use the functions, but that's not very userfriendly imho ;)

What do you think?

bmagyar, 2023-03-14:
Absolutely! I wonder if we could also hook `control_toolbox` into the control.ros.org doxygen generation

christophfroehlich, 2023-03-14:
Yeah, why not. But maybe it's more important to have some high-level human-readable docs on control.ros.org than the API. The latter one is not that easy to find, and I never found something relevant via google but only ROS 1-related on http://docs.ros.org. Probably we should cross-link https://control.ros.org/master/doc/api/ more often.

christophfroehlich, 2024-04-20:
> Absolutely! I wonder if we could also hook `control_toolbox` into the control.ros.org doxygen generation

Will be added with https://github.com/ros-controls/control.ros.org/pull/278