ros-controls/control_toolbox #156: Remove `prefix_is_for_params` from PID

bmagyar, 2023-04-04:


christophfroehlich, 2025-05-23:
quoting @bmagyar 
> yes, I think the idea was to move away from that parameter and maybe even entirely disable that functionality or have one default and not allow different ones

destogl, 2025-07-16:
I agree on "no string manipulation" as this is the issue here. We were removing "/" from parameters and then the interface names of the following controllers that habe prefix with "/" in it were suddenly invalidated.

If we use the local topic names, it means with the node prefix "~/", do we require parameter prefix? I am not sure if we need this for embedding PID to other controllers

closed 2025-07-22 by christophfroehlich

ViktorCVS, 2025-07-23:
@christophfroehlich Since this issue has been resolved, do you think it’s a good time to make those changes?
https://github.com/ros-controls/ros2_controllers/pull/1585#issuecomment-2985693108

christophfroehlich, 2025-07-23:
yess, please!

PR #431 (2025-07-22): Deprecate `prefix_is_for_params` of PidROS
See https://github.com/ros-controls/control_toolbox/issues/156#issuecomment-2985605034

Fixes #156
PR #434 (2025-07-22): Deprecate `prefix_is_for_params` of PidROS (backport #431)
See https://github.com/ros-controls/control_toolbox/issues/156#issuecomment-2985605034

Fixes #156 <hr>This is an automatic backport of pull request #431 done by [Mergify](https://mergify.com).