ros-controls/control_toolbox #164: LowPassFilter accepts parameters only from node

roncapat, 2023-10-09:
I wanted to use LowPassFilter, but it seems to me that there is no way to pass parameters via API (like `PID` constructor or `setGains()` call).

Wouldn't be better to make `LowPassFilter` support parameter passing via API and a corresponding `LowPassFilterROS` querying parameters from configuration instead? 
Or just modifying LowPassFilter to support both scenarios... 

I would contribute a PR following your feedbacks of course.

roncapat, 2023-10-09:
Ok, I explored a bit the base class code, and I see that references for internal parameters can be obtained through `getParam(name,reference)` call, and the reason why we have a ParameterListener inside this implementation is that the `filter::FilterBase` class is implemented that way. I now wonder only whether there is an overhead in refereshing and querying periodically parameters or not. Can someone point me to any performance evaluation, if ever done in the past?

CalaW, 2024-02-25:
I think it is possible to change the parameters at runtime using `this->set_parameter`. Similar to https://github.com/ros-controls/control_toolbox/blob/759d95422893e03fdc88bfe0430370b13902b6f7/test/control_filters/test_low_pass_filter.cpp#L27

christophfroehlich, 2024-02-25:
I support the idea of a base implementation without ROS, and then being wrapped in the filters-class with the ROS node interfaces.
@roncapat Even if your idea is a few months old, I'd welcome a PR.

PR #222 (2024-11-06): Add standalone version of LPF
Replaces #192 and closes #164

Changes on top:

- I changed the base class to be independent of `filters`
- Used the old class/file name and moved the new one to `control_toolbox` namespace.

The filter plugins should still work as before without any changes (see the test files).
closed 2024-11-06 by christophfroehlich