ros-controls/control_toolbox #205: fatal error LNK1181: missing 'low_pass_filter.lib'

zchengw, 2024-06-24:
I tried to build control_toolbox on Windows (MSVC) (using `colcon build`)
It gives me this LNK1181 error.
```
Building Custom Rule C:/Program Files/ros2_ws/src/control_toolbox/CMakeLists.txt
test_load_low_pass_filter.cpp                                              
LINK : fatal error LNK1181: �޷��������ļ���Release\low_pass_filter.lib��[C:\ProgramFiles\ros2_ws\build\control_toolbox\test_load_low_pass_filter.vcxproj]
Building Custom Rule C:/Program Files/ros2_ws/src/control_toolbox/CMakeLists.txt                                                                                                               
test_low_pass_filter.cpp
[Processing: control_toolbox]                                                                          
LINK : fatal error LNK1181: �޷��������ļ���Release\low_pass_filter.lib��[C:\ProgramFiles\ros2_ws\build\control_toolbox\test_low_pass_filter.vcxproj]                                                                                                           
Failed   <<< control_toolbox [1min 3s, exited with code 1]   
```
Anyone knows how to fix this?

christophfroehlich, 2024-07-15:
symbols are not exported on windows per default.

The filters are not adapted for windows. One solution is to use visibility macros as we have them in the demos for example
https://github.com/ros-controls/ros2_control_demos/blob/master/example_1/hardware/include/ros2_control_demo_example_1/visibility_control.h

If you have fixed this, feel free to open a PR in this repo.

christophfroehlich, 2025-01-21:
should be fixed with #258

closed 2025-01-21 by christophfroehlich