ros-controls/control_toolbox #259: Apt install version has old state

KevVincent, 2025-01-02:
Hey I tried installing the repo using apt and i am getting the error  while building my package that depends on control_toolbox

`opt/ros/jazzy/include/realtime_tools/realtime_tools/realtime_buffer.h:25:2: error: #warning "This header include is deprecated. Please update your code to use 'realtime_buffer.hpp' header." [-Werror=cpp]
   25 | #warning \
      |  ^~~~~~~
In file included from[image]:46:
/opt/ros/jazzy/include/realtime_tools/realtime_tools/realtime_publisher.h:25:2: error: #warning "This header include is deprecated. Please update your code to use 'realtime_publisher.hpp' header." [-Werror=cpp]
   25 | #warning \
      |  ^~~~~~~
`

if i include /opt/ros/jazzy/include/control_toolbox/control_toolbox/pid_ros.hpp:45.

I see that the change was made two weeks ago https://github.com/ros-controls/control_toolbox/pull/247 but the apt install version's pid_control.hpp

[image]

Any help would be appreciated. Thanks in advance.

BR,
Kevin

bmagyar, 2025-01-02:
> ^~~~~~~ In file included from![Screenshot from 2025-01-02 15-36-13](

Worry not. Releases don't make themselves.

When in doubt whether some change should be expected to be in the latest
apt release or not, check the repo history for a version number. The change
you are looking for hasn't been released yet as it is above the last
version tag in the history.

https://github.com/ros-controls/control_toolbox/commits/ros2-master/

christophfroehlich, 2025-01-02:
This means you can either

1. wait for the next release and sync of Jazzy (one or more month, as the latest has been a couple of days ago)
2. compile the current HEAD of this repository from source

christophfroehlich, 2025-01-20:
should be solved with [last night's sync](https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2025-01-20/41633/1)

closed 2025-01-20 by christophfroehlich