ros-controls/control_toolbox #274: Docker build using humble branch failing

Abhishek8857, 2025-01-19:
'302.9 Starting >>> control_toolbox
312.7 --- stderr: control_toolbox
312.7 In file included from /colcon_ws/src/control_toolbox/src/control_filters/low_pass_filter.cpp:15:
312.7 /colcon_ws/src/control_toolbox/include/control_filters/low_pass_filter.hpp:25:10: fatal error: control_toolbox/low_pass_filter_parameters.hpp: No such file or directory
312.7    25 | #include "control_toolbox/low_pass_filter_parameters.hpp"
312.7       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
312.7 compilation terminated.
312.7 gmake[2]: *** [CMakeFiles/low_pass_filter.dir/build.make:76: CMakeFiles/low_pass_filter.dir/src/control_filters/low_pass_filter.cpp.o] Error 1
312.7 gmake[1]: *** [CMakeFiles/Makefile2:258: CMakeFiles/low_pass_filter.dir/all] Error 2
312.7 gmake[1]: *** Waiting for unfinished jobs....
312.7 gmake: *** [Makefile:146: all] Error 2
312.7 ---
312.7 Failed   <<< control_toolbox [9.80s, exited with code 2]
312.8 
312.8 Summary: 18 packages finished [5min 12s]
312.8   1 package failed: control_toolbox'

christophfroehlich, 2025-01-19:
will fix itself with the upcoming humble sync. for now, you can compile the latest version of generate_parameter_library from source

christophfroehlich, 2025-01-21:
This should be fixed now with https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2025-01-21/41658

closed 2025-01-21 by christophfroehlich