ros-controls/control_toolbox #275: Misiing file exponential_filter_parameters.hpp

rb19782, 2025-01-19:
Hello,
During building on Jazzy I have following error:
In file included from /home/pi/ros2_ws/src/control_toolbox/src/control_filters/exponential_filter.cpp:15:
/home/pi/ros2_ws/src/control_toolbox/include/control_filters/exponential_filter.hpp:25:10: fatal error: control_toolbox/exponential_filter_parameters.hpp: No such file or directory
   25 | #include "control_toolbox/exponential_filter_parameters.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

WHree is a problem?
Rafal

christophfroehlich, 2025-01-20:
You need generate_parameter_library 0.4.0. This should be solved with the [jazzy sync last night](https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2025-01-20/41633/1). Can you confirm that?

rb19782, 2025-01-20:
It's working now.
During bulding full demo repository many warnings, but no errors. Thanks :)

closed 2025-01-20 by rb19782