ros-controls/control_toolbox #284: PidROS (and PID) zeroing command when error remains constant

mpollayil, 2025-02-10:
I'm on ROS2 Jazzy and I'm experiencing the following issue when trying to use a `control_toolbox::PidROS` in a ros2 controller. The computed command is randomly going to `0.0.` even when the error is not `0.0`. I even only have the proportional gain equal to `1.0` and both the derivative and integral gains are `0.0`.

I'm trying to debug by plotting the `control_msgs/PidState`. Here is the result:

[image]

The derivative and integral terms should not be influencing anything here since their gains are set to `0.0`. Also the anti-windup is set to `false`. The error is finite and the `dt` I use is the controllers `update` functions input `const rclcpp::Duration& period`.

I'm not understanding what's going on here! Any help would be appreciated as this seems to be a big bug? Am I doing something wrong?

mpollayil, 2025-02-10:
Hey @saikishor,

Thank you for the prompt response. I'm pretty sure that `dt_s ` going to zero is the first thing I checked. Tomorrow morning I'll plot it and see again.

Is there anything else that I might be missing that makes the command go to zero?

saikishor, 2025-02-10:
> Is there anything else that I might be missing that makes the command go to zero?

AFAIK, nothing else

christophfroehlich, 2025-02-10:
Which method do you use to update the PID? `compute_command(error, dt_s)` or `compute_command(error, error_dot, dt_s)`?

mpollayil, 2025-02-11:
So, I apologize for a wrong info I gave earlier: the `dt_s` seems to be actually going to 0.

[image]

saikishor, 2025-02-11:
This is weird are you saying that the period received by the controller is zero?

Your fix of hardcoded period works?

mpollayil, 2025-02-11:
> This is weird are you saying that the period received by the controller is zero?

Yes, this is actually very weird. The controller's `dt_s` seems to be oscillating and going to zero almost periodically.

> Your fix of hardcoded period works?

No! :( Even though I'm passing a fixed duration to the `compute_command`, the internal `dt_s` seems oscillating? Am I getting this wrong?

mpollayil, 2025-02-11:
Ok, so there is something weird happening with my code it seems. I'll get back to you soon and update more in detail.

mpollayil, 2025-02-11:
Ok, so guys. The ros2 controller was giving exactly `0.0` as controller period and this was what I was initially using ad the `dt_s` of the `compute_period`. I never expected this to happen. So, checking the value of that period and modifying it to be above a lower threshold solved the issue with `PidROS`.

I apologize for the confusion. Thank yo again for your prompt support!

christophfroehlich, 2025-02-11:
where does the 0 come from? Passed from the controller manager to the update method?

mpollayil, 2025-02-11:
Yes, it is passed from the controller manager to my ros2 controller's update method. Isn't this strange? I should say that I did this test in simulation with Gazebo and the real-time factor is not so great... Not sure if that's causing the 0 control period.