ros-controls/control_toolbox #287: Merge JointSaturationLimiter and RateLimiter

christophfroehlich, 2025-02-13:
We have a [`JointSaturationLimiter`](https://github.com/ros-controls/ros2_control/blob/master/joint_limits/src/joint_saturation_limiter.cpp) that limits positions, velocities, and acceleration for the arms. Can we reuse that logic also here, by extending the limiter with a new `on_enforce` method to also accept the `Twist` (or `TwistStamped`) messages and do the limiting there. I suppose we can reuse the logic, and then we hopefully don't need to have two implementations. AFAIK, we can even create a new class for it as `Joint` in the name implies to be used on joint only. 

 _Originally posted by @destogl in [#1346](https://github.com/ros-controls/ros2_controllers/issues/1346#issuecomment-2573760791)_

christophfroehlich, 2025-02-13:
@tpoignonec as you have contributed to the RateLimiter, have you ever considered and analyzed the JointSaturationLimiter? Maybe we have a smarter approach there, and could move it to control_toolbox package.

tpoignonec, 2025-02-15:
No I never checked out `JointSaturationLimiter`, but I don't mind taking a look on Monday, it does seem like the functionalities have much in commun

tpoignonec, 2025-02-18:
@christophfroehlich The algorithm in [JointSaturationLimiter](https://github.com/ros-controls/ros2_control/blob/master/joint_limits/src/joint_saturation_limiter.cpp) seems much more robust, especially the mechanism to check predicted position limits for the next sampling time as well as the stopping feasibility w.r.t. max (de)acceleration.
I don't see any reason to not merge either, keeping the logic from `JointSaturationLimiter`, but the syntax and interface from `RateLimiter`.

Actually, would it be sufficient to simply extend the `RateLimiter` functionalities and to use them with the current API ?
(As opposed to directly providing `on_enforce`-like methods in the control toolbox)
The cost of wrapping the the `RateLimiter` for joints or twist is pretty limited, but I don't think there is a unique "good" way to do that for instance for parameters management.
Also, a component could require all rate-limited variables (e.g., joints or Cartesian coordinates) to stay in sync or something, so that if the motion is halted on one joint, it is so for all joints.

christophfroehlich, 2025-03-14:
I think we would have to extract the algorithm (like we did from the SpeedLimiter for the RateLimiter), and use it for joints in the joint_limits package, and for generic variables in the RateLimiter here. It would be good to understand if the other algorithm has better behavior as what we have discussed in https://github.com/ros-controls/control_toolbox/pull/240
then it would be worth it