ros-controls/control_toolbox #291: fatal error: control_toolbox/low_pass_filter_parameters.hpp: No such file or directory

fzoric8, 2025-03-06:
Hi, 

When building `control_toolbox` on the `humble` branch, 

I get following error: 
```
In file included from /root/ros2_ws/src/ros-controls/control_toolbox/src/control_filters/rate_limiter.cpp:15:
/root/ros2_ws/src/ros-controls/control_toolbox/include/control_filters/rate_limiter.hpp:27:10: fatal error: control_toolbox/rate_limiter_parameters.hpp: No such file or directory
   27 | #include "control_toolbox/rate_limiter_parameters.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/rate_limiter.dir/build.make:76: CMakeFiles/rate_limiter.dir/src/control_filters/rate_limiter.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:310: CMakeFiles/rate_limiter.dir/all] Error 2
In file included from /root/ros2_ws/src/ros-controls/control_toolbox/src/control_filters/low_pass_filter.cpp:15:
/root/ros2_ws/src/ros-controls/control_toolbox/include/control_filters/low_pass_filter.hpp:25:10: fatal error: control_toolbox/low_pass_filter_parameters.hpp: No such file or directory
   25 | #include "control_toolbox/low_pass_filter_parameters.hpp"
```

i.e., `control_toolbox` package fails on build. 

Thanks for looking into the issue.

christophfroehlich, 2025-03-06:
Please update generate_parameter_library to the recent version 0.4.0, see https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2025-01-21/41658

closed 2025-03-06 by christophfroehlich

fzoric8, 2025-03-06:
After inspecting `CMakeLists.txt` I've realized I'm missing: `generate_parameter_library`. 

Installed it with: 
```
apt-get install ros-humble-generate-parameter-library
source /opt/ros/humble/
```

christophfroehlich, 2025-03-06:
You haven't had installed it at all? then I wonder why cmake succeed before the compilation started 🤷