ros-controls/control_toolbox #345: Improve member variables of low_pass_filter

christophfroehlich, 2025-04-24:
## Background

Currently, there are two member variables for different templated specialization (`wrench`, and all others use `T`)
https://github.com/ros-controls/control_toolbox/blob/b1f2619ef54a1ba3fdb38cb9dcce1745dd6ac31b/control_toolbox/include/control_toolbox/low_pass_filter.hpp#L126-L129

Improve that by only adding a single variant depending on the template argument.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

pedroazeredo04, 2025-04-26:
Hello :) I've never contributed to this repository before, and I would like to try and tackle this issue

christophfroehlich, 2025-04-26:
Something like this might work with C++17
```c++
#include <type_traits>

// Define the storage type based on T
using StorageType = typename std::conditional<
    std::is_same<T, geometry_msgs::msg::WrenchStamped>::value,
    Eigen::Matrix<double, 6, 1>,
    T
>::type;

// Member variables
StorageType filtered_value_;
StorageType filtered_old_value_;
StorageType old_value_;

if constexpr (std::is_same<T, geometry_msgs::msg::WrenchStamped>::value)
{
  for (Eigen::Index i = 0; i < 6; ++i) {
  filtered_value[i] = filtered_old_value[i] = old_value[i] = std::numeric_limits<double>::quiet_NaN();
  }
}
else
{
  filtered_value = filtered_old_value = old_value = std::numeric_limits<T>::quiet_NaN();
}
```

christophfroehlich, 2025-04-27:
@pedroazeredo04 I know that I proposed the type_traits above first, but after thinking once again: The second approach would be much more scaleable, and we could easily use that for all other filters.

Something like this, plus other necessary member functions like `isnan` or operator overloads for `+operator`, `*operator`, which can go in its own header file:
```
// Default: storage type is just T itself
template<typename T>
struct FilterTraits
{
    using StorageType = T;

    static void initialize(StorageType& storage)
    {
        storage = std::numeric_limits<T>::quiet_NaN();
    }
};

// Specialization: for WrenchStamped, use Eigen::Matrix
template<>
struct FilterTraits<geometry_msgs::msg::WrenchStamped>
{
    using StorageType = Eigen::Matrix<double, 6, 1>;

    static void initialize(StorageType& storage)
    {
      for (Eigen::Index i = 0; i < 6; ++i)
      {    
        storage[i] = std::numeric_limits<double>::quiet_NaN();
      }
    }
};
```

and then in the filter classes

```
    using Traits = FilterTraits<T>;
    using StorageType = typename Traits::StorageType;
    StorageType filtered_value_, filtered_old_value_, old_value_;

...
        Traits::initialize(filtered_value_);
        Traits::initialize(filtered_old_value_);
        Traits::initialize(old_value_);
```

closed 2025-05-08 by christophfroehlich