ros-controls/control_toolbox #365: Refactor exponential filter

christophfroehlich, 2025-05-08:
## Background

Refactor the exponential filter to have the same structure of the low-pass filter, using the new storage variable template from 

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step 3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [Robotics Stack Exchange](https://robotics.stackexchange.com)

**Good luck with your first issue!**

pedroazeredo04, 2025-05-09:
Hey, if you think it's ok, I would like to continue my work on this!!

But since this is a good first issue and I have already contributed to this repository before, if you think it's better, I could leave this to someone new and try and solve the issue #350

someshkharat3199, 2025-05-10:
I'd like to work on this issue.

pedroazeredo04, 2025-05-10:
@christophfroehlich 
> do you have a suggestion here how we harmonize the two filters (low-pass and exponential), because the latter one has only a single-method base implementation? should we extract the same from the low-pass filter, does this make sense?

Humm, I think that would make sense, but then we wouldn't need the file [`include/control_toolbox/low_pass_filter.hpp`](https://github.com/ros-controls/control_toolbox/blob/ros2-master/control_toolbox/include/control_toolbox/low_pass_filter.hpp) at all, right? Because we would just use the single-method implementation in the file [`include/control_filters/low_pass_filter.hpp`](https://github.com/ros-controls/control_toolbox/blob/ros2-master/control_toolbox/include/control_filters/low_pass_filter.hpp), and have the class `control_filters::LowPassFilter` manage the previous measurement values and whatnot, just like it is done with the `control_filters::ExponentialFilter` class, in the file [`include/control_filters/exponential_filter.hpp`](https://github.com/ros-controls/control_toolbox/blob/ros2-master/control_toolbox/include/control_filters/exponential_filter.hpp)

I think to harmonize the two filters, we have to decide what makes more sense: To have the responsibility of managing the previous values to be in the class `control_toolbox::LowPassFilter` or in the class `control_filters::LowPassFilter`. If we choose the second option, then I think it makes sense to extract the same single-method base implementation from the low-pass filter.

Is that what you were asking? Or I understood something wrong

silanus23, 2025-10-01:
@christophfroehlich I've opened a pull request to address this issue. Could you please take a look when you have a moment? #493